dawidchyrzynski/arduino-home-assistant #230: Missing checks if library is used in a wrong way?

klausj1, 2024-02-10:
Hi,

Used library version: 2.0
Board: eps32C3 dev module

I had two issues which took me some time to find out:

- The uniqueID of a device type must not contain a space
- WifiClient and HAMqtt must be defined in the code before the entities are defined

I did not see an error message, it just did not work.

I am not sure if there are error messages already, because I did not manage to change the "Core Debug Level:" in the board configuration - I never saw any message, also not from other libraries.

If you are interested, I can also provide a more detailed description.

dawidchyrzynski, 2024-02-10:
Hi @klausj1,

I will consider adding additional checks, possibly enabled by a macro; however, I am not fan of allocating resources (RAM/flash) for such scenarios. My general recommendation is to thoroughly review the documentation, as it describes your findings.

https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/introduction.html
https://dawidchyrzynski.github.io/arduino-home-assistant/documents/library/compiler-macros.html

klausj1, 2024-02-11:
OK, Thx. Feel free to close this issue ;)

closed 2024-05-19 by dawidchyrzynski