dawidchyrzynski/arduino-home-assistant #234: Availability topic doesn't change with data topic

f-hnt, 2024-02-16:
Availability topic doesn't change with data topic and stays at the default "aha".

By setting the data topic in the setup section with:
mqtt.setDataPrefix("DataTopic");
the availability of the device is being reported in "aha" topic. States, data, etc. are correctly reported in "DataTopic"

Is this a normal behavior?
I expected after changing the data data topic, the MQTT structure to be like that:

Data:
DataTopic/UniqueID/switch1/....

Availability:
DataTopic/UniqueID/avty_t

dawidchyrzynski, 2024-05-19:
Hi @f-hnt,

The `setDataPrefix` method must be called before `enableSharedAvailability`. Please let me know if it solves your problem.

f-hnt, 2024-05-19:
It worked. Thank you a lot :)

closed 2024-05-19 by dawidchyrzynski