dawidchyrzynski/arduino-home-assistant #235: HADeviceTrigger not working, no activity in HomeAssistat

mariusmotea, 2024-02-20:
Hi,

I try to have some hardware switches working with HomeAssistant but unfortunately they are not working. If i add other types like switches, buttons, sensors they show in HomeAssistant right away. I turn debug mode on for the MQTT integration and i can only see some entries when the ESP32S3 boot and connect but nothing when i press the buttons. I test with the example "[multi-state-button](https://github.com/dawidchyrzynski/arduino-home-assistant/tree/main/examples/multi-state-button)" using ethernet W5500 module and WiFi, no change. With MQTT explorer i can see activity on the topic used by this integration, but no reaction in HomeAssistant and no entity is being created (like is happening with HAButton, HASwitch, etc). I use the last release of HomeAssistant and i believe some changes may exist at the api level.

dawidchyrzynski, 2024-02-23:
Hi @mariusmotea,

Triggers are not represented as entities in Home Assistant. They can only be utilized within automations. When creating a new automation, please select your device. In the "Trigger" dropdown, you will find a list of triggers exposed by the selected device.

https://www.home-assistant.io/docs/automation/trigger/

mariusmotea, 2024-02-25:
Thanks @dawidchyrzynski, i missed this info. It is working good.

closed 2024-02-25 by mariusmotea