dawidchyrzynski/arduino-home-assistant #243: Example for dynamic configurationUrl based on current IP

Routhinator, 2024-03-12:
I've been trying to set the configuraition URL on startup dynamically, however the only working code I've managed to get to compile results in a bytes object instead of a string, which breaks the HA integration. 

Can someone add an example on how to achieve this?

```c++
// Setup vars
String localIP;
const char* url;

...
  // In Setup
  // Set device as a Wi-Fi Station
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to Wi-Fi..");
  }
  localIP = WiFi.localIP().toString();

  Serial.print("Sensor IP Address: ");
  Serial.println(localIP);
  Serial.println();

  url = String(String("http://") + localIP).c_str();
  device.setConfigurationUrl(String(url).c_str());
```

This results in the following in the MQTT messages:

```json
{"cu":"��?"}
```

Unfortunately I've not found a way to combine an `IPAddress` variable with a const like `"http://"` without all the `String().c_str()` conversions, and from the `Serial.println` I can verify the IP is properly set in the string...

Routhinator, 2024-03-13:
Ok I finally got something that works. Took some fiddling around, I need to refresh my understanding of var types and pointers in C++ it seems.

```c++
// Setup vars
String localIP;
char url[20];

...
void setup() {
...
  // Set device as a Wi-Fi Station
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to Wi-Fi..");
  }
  localIP = WiFi.localIP().toString();

  Serial.print("Sensor IP Address: ");
  Serial.println(localIP);
  Serial.println();
  
   sprintf(url, "http://%s", localIP.c_str());
   Serial.println(url);
   device.setConfigurationUrl(url);

...
}

```

Which produces the proper configuration URL value:

```json
{
...
  "cu": "http://192.168.1.99"
...
}
```

romoloman, 2024-05-19:
This works too
```
char c_url[25];
String url="http://";
url.concat(WiFi.localIP().toString());
strcpy(c_url,url.c_str());
device.setConfigurationUrl(c_url);
```
but 
```
String url="http://";
url.concat(WiFi.localIP().toString());
device.setConfigurationUrl(url.c_str());
```
doesn't

jellewie, 2025-12-29:
I always have problems with the URL command; it keeps giving HA warnings

Config
```
  String URL = "http://" + IpAddress2String(WiFi.localIP());
  static String URL = "http://192.168.50.205";
  static char configUrl[30];  // Adjust size as needed, large enough to hold the URL
  URL.toCharArray(configUrl, sizeof(configUrl));
  device.setConfigurationUrl(configUrl);
```

Results in
```
Logger: homeassistant.components.mqtt.entity
Source: components/mqtt/entity.py:159
integration: MQTT (documentation, issues)
First occurred: 28 December 2025 at 13:30:30 (1 occurrence)
Last logged: 28 December 2025 at 13:30:30

Error 'invalid url for dictionary value @ data['device']['configuration_url']' when processing MQTT discovery message topic: 'homeassistant/light/0010fa6e384a/smartStair/config', message: '{'name': 'stairenabled', 'state_topic': 'aha/0010fa6e384a/smartStair/stat_t', 'device': {'identifiers': '0010fa6e384a', 'manufacturer': 'JelleWho', 'sw_version': '1.0', 'configuration_url': '192.168.50.213', 'name': 'smartStair', 'model': 'ESP32 Stair'}, 'unique_id': 'smartStair', 'command_topic': 'aha/0010fa6e384a/smartStair/cmd_t'}'
```