dawidchyrzynski/arduino-home-assistant #247: Not discovered?

Ramdough, 2024-05-06:
Ok, so I am a bit of a newby….but please help.   

I have copied your code directly but updated the MAC address, IP number, and unique ID (so it stood out more).   My code compiles and writes to the arduino successfully.  

I can see my arduino and pig it on the network.   

Under MQTT on HA, I am able to publish to myself using the manual publish and listen feature.  I have listen set to #, so I am listening to all traffic.  So mosquito appears to be working.  Discovery is on.  

For some reason, I do not get a new sensor created using this code.  

What should I try next?

Ramdough, 2024-05-08:
Any thoughts anyone?

jwilleke, 2024-05-09:
Got any code to share?

Ramdough, 2024-05-10:
> Got any code to share?

I used this exact binary-sensor example code but changed the MAC address, the unique ID, and the broker IP address.

Ramdough, 2024-05-10:
I am using an arduino mega and plan to modify this code to read all dio pins.

romoloman, 2024-05-18:
Same on esp32 binary sensor is not discovered but is published on MQTT... (checking with MQTT explorer)

dawidchyrzynski, 2024-05-19:
Hi @Ramdough, which version of Home Assistant do you use?

romoloman, 2024-05-19:
With version 2.1.0 everything is fine on HA 2024.5.4, but still persist the issue of multiple devices of the same type in the network, so I had to use the develop branch that implements enableExtendedUniqueIds() using the mac address of the esp 32 as device id.

Ramdough, 2024-05-21:
> Hi @Ramdough, which version of Home Assistant do you use?

I have auto updates so I am always on the latest version.

Ramdough, 2024-05-21:
Thanks everyone for helping!  I appreciate the support.

proasnet, 2024-06-28:
Dear all,
I have the same problem. On my project with ESP32, I have a Wifi connection or Ethernet connection (W5500), switching by user, so, ESP has two connections, two clients.
I have a two constructors 
```
HADevice device;
HAMqtt wifi_mqtt( wificlient , device , 20 );
HAMqtt ethernet_mqtt( ethernetclient , device , 20 );
```
If I disable ethernet
```
HADevice device;
HAMqtt wifi_mqtt( wificlient , device , 20 );
//HAMqtt ethernet_mqtt( ethernetclient , device , 20 );
```
Discovery working.

How can I fix this for both instances?
Thank you very much.

closed 2025-07-06 by dawidchyrzynski

ethanhjennings, 2025-12-14:
As a sanity check for people, if you're using the default MQTT Mosquito add-on it now requires all devices have a username and password to connect. So you need to add a login in the configuration tab for it, and then in your code change the `mqtt.begin()` method call to pass in that username and password:
```
mqtt.begin(BROKER_ADDR, "username", "password");
```

This fixed discovery for me!