dawidchyrzynski/arduino-home-assistant #275: HVAC thermostat is not controllable from HASS

andrewhack, 2024-12-04:
Hi,
Am I missing something - can't control the target temperature from HASS.
Changing the mode is working fine.
I am using the provided HVAC example with Arduino Mega and Ethernet shield.

These are the features which I am using
```
HAHVAC::TargetTemperatureFeature | HAHVAC::PowerFeature | HAHVAC::ModesFeature
```

[image]

andrewhack, 2024-12-04:
My investigation shows that mqtt "mode" and "target temp" entries are missing, just "current temp" is available.
I assume there is no initial or not complete mqtt setup.

Can somebody assist here?

MarshallVisions, 2024-12-16:
am having the same problem... find a solution by any chance?

simbycube, 2025-01-04:
same problem for me too.

MarshallVisions, 2025-01-13:
I seem to have gotten it working with the following example code. I think I needed to delete my old device with my actual thermostat code for current temperature to also start working. Probably because I think they were both getting the same uniqueid. Will do some more investigating.

```
#include <WiFi.h>
#include <ArduinoHA.h>

#define mqttServer IPAddress(192, 168, 0, 126)
const int mqttPort = 1883;
const char *mqttUser = "MQTT-User";
const char *mqttPassword = "3t3rn!tY";
#define WIFI_SSID       "MV-Terminal_2.4GHz"
#define WIFI_PASSWORD   "3t3rn!tY"
// #include <WiFiClient.h>

WiFiClient client;
HADevice device;
HAMqtt mqtt(client, device);
// 
// By default HAHVAC supports only reporting of the temperature.
// You can enable feature you need using the second argument of the constructor.
// Please check the documentation of the HAHVAC class.
HAHVAC hvac(
  "hvac-test",
  HAHVAC::TargetTemperatureFeature | HAHVAC::DefaultFeatures
);

unsigned long lastTempPublishAt = 0;
float lastTemp = 0;

void onTargetTemperatureCommand(HANumeric temperature, HAHVAC* sender) {
    float temperatureFloat = temperature.toFloat();

    Serial.print("Target temperature: ");
    Serial.println(temperatureFloat);

    sender->setTargetTemperature(temperature); // report target temperature back to the HA panel
}

void onPowerCommand(bool state, HAHVAC* sender) {
  if (state) {
    Serial.println("Power on");
  } else {
    Serial.println("Power off");
  }
}

void onModeCommand(HAHVAC::Mode mode, HAHVAC* sender) {
    Serial.print("Mode: ");
    if (mode == HAHVAC::OffMode) {
        Serial.println("off");
    } else if (mode == HAHVAC::AutoMode) {
        Serial.println("auto");
    } else if (mode == HAHVAC::CoolMode) {
        Serial.println("cool");
    } else if (mode == HAHVAC::HeatMode) {
        Serial.println("heat");
    } else if (mode == HAHVAC::DryMode) {
        Serial.println("dry");
    } else if (mode == HAHVAC::FanOnlyMode) {
        Serial.println("fan only");
    }

    sender->setMode(mode); // report mode back to the HA panel
}

void setup() {
    Serial.begin(9600);
    Serial.println("Starting...");

    // Unique ID must be set!
    byte mac[6];
    WiFi.macAddress(mac);
    device.setUniqueId(mac, sizeof(mac));

    // connect to wifi
    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(500); // waiting for the connection
    }
    Serial.println();
    Serial.println("Connected to the network");

    // set device's details (optional)
    device.setName("NodeMCU");
    device.setSoftwareVersion("1.0.0");

    // assign callbacks (optional)
    hvac.onTargetTemperatureCommand(onTargetTemperatureCommand);
    hvac.onPowerCommand(onPowerCommand);
    hvac.onModeCommand(onModeCommand);

    // configure HVAC (optional)
    hvac.setName("My HVAC");
    hvac.setMinTemp(10);
    hvac.setMaxTemp(30);
    hvac.setTempStep(0.5);

    // You can set retain flag for the HA commands
    // hvac.setRetain(true);

    // You can choose which modes should be available in the HA panel
    // hvac.setModes(HAHVAC::OffMode | HAHVAC::CoolMode);

  mqtt.begin(mqttServer, mqttPort, mqttUser, mqttPassword);

}
void loop() {
    mqtt.loop();

    if ((millis() - lastTempPublishAt) > 3000) {
        hvac.setCurrentTemperature(lastTemp);
        lastTempPublishAt = millis();
        lastTemp += 0.5;
    }
}
```

shotreal, 2025-01-18:
same issue here. If you publish something to the [...]/temp_cmd_t topic then itll work

MarshallVisions, 2025-01-19:
still not sure how I fixed it, but the following code works for me for being able to  set the 'Target Temperature' and also show the 'Current Temperature'. Hope this helps someone. It's a retrofitted thermostat which fits on top of my dumb apartment thermostat and uses a servo to move the thermostat adjustment arm. I think alot of the stuff in the Setup() function isn't necessary but I have way too much ADHD to sift through it before posting this lolol.

#include <Arduino.h>
#include <WiFi.h>
#include <ESP32Servo.h>
#include <DHT.h>
#include <ArduinoHA.h>

// WiFi Settings
const char *ssid = "MV-Terminal_2.4GHz";
const char *password = "3t3rn!tY";
IPAddress local_IP(192, 168, 0, 215);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);

// MQTT Settings
#define mqttServer IPAddress(192, 168, 0, 126)
// const int mqttPort = 1883;
const char *mqttUser = "MQTT-User";
const char *mqttPassword = "3t3rn!tY";

// DHT11 Settings
#define DHTPIN 4 // DHT11 data pin
#define DHTTYPE DHT11
DHT dht(DHTPIN, DHTTYPE);
float dhtOffset = -5;
float dhtLastCheck;
float dhtCheckInterval = 1000;

// Home Assistant Settings
// Entity Settings
WiFiClient wifiClient;
HADevice device("RetroStat-Basic");
HAMqtt mqtt(wifiClient, device);
HAHVAC climate(
    "retrostat-hvac", HAHVAC::TargetTemperatureFeature);

HANumber humidity("RetroStat-Basic-Humidity");
HANumber dhtOffsetNumber("RetroStat-Basic-DHT-Offset");

// Servo Settings
Servo servo;
const int servoPin = 7;

// Temperature Settings
float currentTemperature = 20; // Initial temperature set to 20 degrees
float targetTemperature = 20;
const float minTemp = 10;
const float maxTemp = 35;

// Function declarations
void setServoPosition(int temp);
void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender);
void onDhtOffsetCommand(HANumeric value, HANumber *sender);

void setup()
{
    Serial.begin(115200);
    dht.begin();
    servo.attach(servoPin, 500, 2500);

    // Connect to WiFi
    Serial.println("Connecting to WiFi...");
    // WiFi.config(local_IP, gateway, subnet);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }
    Serial.println("WiFi connected");

    // Connect to MQTT
    Serial.println("Connecting to MQTT...");
    mqtt.begin(mqttServer, 1883, mqttUser, mqttPassword);
    mqtt.setDataPrefix("homeassistant/climate"); // not fully sure about this one. defaults to "aha"

    // Initialize Home Assistant components
    device.setName("RetroStat Basic");
    device.setSoftwareVersion("2.0.0");
    device.setManufacturer("MarshallVisions");
    device.setModel("RS-Basic");
    device.enableSharedAvailability();
    device.enableLastWill();
    climate.setName("RetroStat");
    climate.setMinTemp(minTemp);
    climate.setMaxTemp(maxTemp);
    climate.setCurrentTemperature(currentTemperature);
    climate.setTargetTemperature(targetTemperature);
    climate.setRetain(true);
    climate.onTargetTemperatureCommand(onTargetTemperatureCommand);
    humidity.setName("Humidity");
    humidity.setRetain(true);
    dhtOffsetNumber.setName("DHT Temperature Offset");
    dhtOffsetNumber.setRetain(true);
    dhtOffsetNumber.onCommand(onDhtOffsetCommand);

    Serial.println("Setup complete.");
}

void loop()
{
    mqtt.loop();

    // Read temperature from DHT11
    float temp = dht.readTemperature() + dhtOffset;
    float hum = dht.readHumidity();
    if (millis() - dhtLastCheck > dhtCheckInterval)
    {
        currentTemperature = temp;
        climate.setCurrentTemperature((float)currentTemperature);
        humidity.setState(hum);
        dhtLastCheck = millis();
        Serial.print("Current temperature from DHT11: ");
        Serial.println(temp);
        Serial.print("Current humidity from DHT11: ");
        Serial.println(hum);
    }
}

void setServoPosition(int temp)
{
    int pos = map(temp, minTemp, maxTemp, 180, 0);
    Serial.print("Setting servo position to: ");
    Serial.println(pos);
    servo.write(pos);
}

// Function to handle target temperature command from Home Assistant
void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender)
{
    float temperatureFloat = temperature.toFloat();
    if (temperatureFloat >= minTemp && temperatureFloat <= maxTemp)
    {
        targetTemperature = temperatureFloat;
        Serial.print("Target temperature command received. New target temperature: ");
        Serial.println(temperatureFloat);
        setServoPosition(targetTemperature);
        sender->setTargetTemperature(temperature); // report target temperature back to the HA panel
    }
}

void onDhtOffsetCommand(HANumeric value, HANumber *sender)
{
    float offsetFloat = value.toFloat();
    if (offsetFloat >= -10 && offsetFloat <= 10)
    {
        dhtOffset = offsetFloat;
        Serial.print("DHT value command received. New value: ");
        Serial.println(offsetFloat);
        dhtLastCheck = millis();
        sender->setState(dhtOffset);
    }
}

MarshallVisions, 2025-01-19:
still not sure how I fixed it, but the following code works for me for being able to set the 'Target Temperature' and also show the 'Current Temperature'. Hope this helps someone. It's a retrofitted thermostat which fits on top of my dumb apartment thermostat and uses a servo to move the thermostat adjustment arm. I think alot of the stuff in the Setup() function isn't necessary but I have way too much ADHD to sift through it before posting this lolol.

#include <Arduino.h>
#include <WiFi.h>
#include <ESP32Servo.h>
#include <DHT.h>
#include <ArduinoHA.h>

// WiFi Settings
const char *ssid = "MV-Terminal_2.4GHz";
const char *password = "3t3rn!tY";
IPAddress local_IP(192, 168, 0, 215);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);

// MQTT Settings
#define mqttServer IPAddress(192, 168, 0, 126)
// const int mqttPort = 1883;
const char *mqttUser = "MQTT-User";
const char *mqttPassword = "**********";

// DHT11 Settings
#define DHTPIN 4 // DHT11 data pin
#define DHTTYPE DHT11
DHT dht(DHTPIN, DHTTYPE);
float dhtOffset = -5;
float dhtLastCheck;
float dhtCheckInterval = 1000;

// Home Assistant Settings
// Entity Settings
WiFiClient wifiClient;
HADevice device("RetroStat-Basic");
HAMqtt mqtt(wifiClient, device);
HAHVAC climate(
    "retrostat-hvac", HAHVAC::TargetTemperatureFeature);

HANumber humidity("RetroStat-Basic-Humidity");
HANumber dhtOffsetNumber("RetroStat-Basic-DHT-Offset");

// Servo Settings
Servo servo;
const int servoPin = 7;

// Temperature Settings
float currentTemperature = 20; // Initial temperature set to 20 degrees
float targetTemperature = 20;
const float minTemp = 10;
const float maxTemp = 35;

// Function declarations
void setServoPosition(int temp);
void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender);
void onDhtOffsetCommand(HANumeric value, HANumber *sender);

void setup()
{
    Serial.begin(115200);
    dht.begin();
    servo.attach(servoPin, 500, 2500);

    // Connect to WiFi
    Serial.println("Connecting to WiFi...");
    // WiFi.config(local_IP, gateway, subnet);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }
    Serial.println("WiFi connected");

    // Connect to MQTT
    Serial.println("Connecting to MQTT...");
    mqtt.begin(mqttServer, 1883, mqttUser, mqttPassword);
    mqtt.setDataPrefix("homeassistant/climate"); // not fully sure about this one. defaults to "aha"

    // Initialize Home Assistant components
    device.setName("RetroStat Basic");
    device.setSoftwareVersion("2.0.0");
    device.setManufacturer("MarshallVisions");
    device.setModel("RS-Basic");
    device.enableSharedAvailability();
    device.enableLastWill();
    climate.setName("RetroStat");
    climate.setMinTemp(minTemp);
    climate.setMaxTemp(maxTemp);
    climate.setCurrentTemperature(currentTemperature);
    climate.setTargetTemperature(targetTemperature);
    climate.setRetain(true);
    climate.onTargetTemperatureCommand(onTargetTemperatureCommand);
    humidity.setName("Humidity");
    humidity.setRetain(true);
    dhtOffsetNumber.setName("DHT Temperature Offset");
    dhtOffsetNumber.setRetain(true);
    dhtOffsetNumber.onCommand(onDhtOffsetCommand);

    Serial.println("Setup complete.");
}

void loop()
{
    mqtt.loop();

    if (millis() - dhtLastCheck > dhtCheckInterval)
    {
        // Read temperature from DHT11
        float temp = dht.readTemperature() + dhtOffset;
        float hum = dht.readHumidity();
        currentTemperature = temp;
        climate.setCurrentTemperature((float)currentTemperature);
        humidity.setState(hum);
        dhtLastCheck = millis();
        Serial.print("Current temperature from DHT11: ");
        Serial.println(temp);
        Serial.print("Current humidity from DHT11: ");
        Serial.println(hum);
    }
}

void setServoPosition(int temp)
{
    int pos = map(temp, minTemp, maxTemp, 180, 0);
    Serial.print("Setting servo position to: ");
    Serial.println(pos);
    servo.write(pos);
}

// Function to handle target temperature command from Home Assistant
void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender)
{
    float temperatureFloat = temperature.toFloat();
    if (temperatureFloat >= minTemp && temperatureFloat <= maxTemp)
    {
        targetTemperature = temperatureFloat;
        Serial.print("Target temperature command received. New target temperature: ");
        Serial.println(temperatureFloat);
        setServoPosition(targetTemperature);
        sender->setTargetTemperature(temperature); // report target temperature back to the HA panel
    }
}

void onDhtOffsetCommand(HANumeric value, HANumber *sender)
{
    float offsetFloat = value.toFloat();
    if (offsetFloat >= -10 && offsetFloat <= 10)
    {
        dhtOffset = offsetFloat;
        Serial.print("DHT value command received. New value: ");
        Serial.println(offsetFloat);
        dhtLastCheck = millis();
        sender->setState(dhtOffset);
    }
}

MarshallVisions, 2025-01-19:
still not sure how I fixed it, but the following code works for me for being able to set the 'Target Temperature' and also show the 'Current Temperature'. Hope this helps someone. It's a retrofitted thermostat which fits on top of my dumb apartment thermostat and uses a servo to move the thermostat adjustment arm. I think alot of the stuff in the Setup() function isn't necessary but I have way too much ADHD to sift through it before posting this lolol.

```
#include <Arduino.h>
#include <WiFi.h>
#include <ESP32Servo.h>
#include <DHT.h>
#include <ArduinoHA.h>

// WiFi Settings
const char *ssid = "MV-Terminal_2.4GHz";
const char *password = "3t3rn!tY";
IPAddress local_IP(192, 168, 0, 215);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);

// MQTT Settings
#define mqttServer IPAddress(192, 168, 0, 126)
// const int mqttPort = 1883;
const char *mqttUser = "MQTT-User";
const char *mqttPassword = "***********";

// DHT11 Settings
#define DHTPIN 4 // DHT11 data pin
#define DHTTYPE DHT11
DHT dht(DHTPIN, DHTTYPE);
float dhtOffset = -5;
float dhtLastCheck;
float dhtCheckInterval = 1000;

// Home Assistant Settings
// Entity Settings
WiFiClient wifiClient;
HADevice device("RetroStat-Basic");
HAMqtt mqtt(wifiClient, device);
HAHVAC climate(
    "retrostat-hvac", HAHVAC::TargetTemperatureFeature);

HANumber humidity("RetroStat-Basic-Humidity");
HANumber dhtOffsetNumber("RetroStat-Basic-DHT-Offset");

// Servo Settings
Servo servo;
const int servoPin = 7;

// Temperature Settings
float currentTemperature = 20; // Initial temperature set to 20 degrees
float targetTemperature = 20;
const float minTemp = 10;
const float maxTemp = 35;

// Function declarations
void setServoPosition(int temp);
void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender);
void onDhtOffsetCommand(HANumeric value, HANumber *sender);

void setup()
{
    Serial.begin(115200);
    dht.begin();
    servo.attach(servoPin, 500, 2500);

    // Connect to WiFi
    Serial.println("Connecting to WiFi...");
    // WiFi.config(local_IP, gateway, subnet);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }
    Serial.println("WiFi connected");

    // Connect to MQTT
    Serial.println("Connecting to MQTT...");
    mqtt.begin(mqttServer, 1883, mqttUser, mqttPassword);
    mqtt.setDataPrefix("homeassistant/climate"); // not fully sure about this one. defaults to "aha"

    // Initialize Home Assistant components
    device.setName("RetroStat Basic");
    device.setSoftwareVersion("2.0.0");
    device.setManufacturer("MarshallVisions");
    device.setModel("RS-Basic");
    device.enableSharedAvailability();
    device.enableLastWill();
    climate.setName("RetroStat");
    climate.setMinTemp(minTemp);
    climate.setMaxTemp(maxTemp);
    climate.setCurrentTemperature(currentTemperature);
    climate.setTargetTemperature(targetTemperature);
    climate.setRetain(true);
    climate.onTargetTemperatureCommand(onTargetTemperatureCommand);
    humidity.setName("Humidity");
    humidity.setRetain(true);
    dhtOffsetNumber.setName("DHT Temperature Offset");
    dhtOffsetNumber.setRetain(true);
    dhtOffsetNumber.onCommand(onDhtOffsetCommand);

    Serial.println("Setup complete.");
}

void loop()
{
    mqtt.loop();

    if (millis() - dhtLastCheck > dhtCheckInterval)
    {
        // Read temperature from DHT11
        float temp = dht.readTemperature() + dhtOffset;
        float hum = dht.readHumidity();
        currentTemperature = temp;
        climate.setCurrentTemperature((float)currentTemperature);
        humidity.setState(hum);
        dhtLastCheck = millis();
        Serial.print("Current temperature from DHT11: ");
        Serial.println(temp);
        Serial.print("Current humidity from DHT11: ");
        Serial.println(hum);
    }
}

void setServoPosition(int temp)
{
    int pos = map(temp, minTemp, maxTemp, 180, 0);
    Serial.print("Setting servo position to: ");
    Serial.println(pos);
    servo.write(pos);
}

// Function to handle target temperature command from Home Assistant
void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender)
{
    float temperatureFloat = temperature.toFloat();
    if (temperatureFloat >= minTemp && temperatureFloat <= maxTemp)
    {
        targetTemperature = temperatureFloat;
        Serial.print("Target temperature command received. New target temperature: ");
        Serial.println(temperatureFloat);
        setServoPosition(targetTemperature);
        sender->setTargetTemperature(temperature); // report target temperature back to the HA panel
    }
}

void onDhtOffsetCommand(HANumeric value, HANumber *sender)
{
    float offsetFloat = value.toFloat();
    if (offsetFloat >= -10 && offsetFloat <= 10)
    {
        dhtOffset = offsetFloat;
        Serial.print("DHT value command received. New value: ");
        Serial.println(offsetFloat);
        dhtLastCheck = millis();
        sender->setState(dhtOffset);
    }
}
```

andrewhack, 2025-01-23:
> still not sure how I fixed it, but the following code works for me for being able to set the 'Target Temperature' and also show the 'Current Temperature'. Hope this helps someone. It's a retrofitted thermostat which fits on top of my dumb apartment thermostat and uses a servo to move the thermostat adjustment arm.
> 
> ```
> #include <Arduino.h>
> #include <WiFi.h>
> #include <ESP32Servo.h>
> #include <DHT.h>
> #include <ArduinoHA.h>
> 
> // WiFi Settings
> const char *ssid = "MV-Terminal_2.4GHz";
> const char *password = "3t3rn!tY";
> IPAddress local_IP(192, 168, 0, 215);
> IPAddress gateway(192, 168, 0, 1);
> IPAddress subnet(255, 255, 255, 0);
> 
> // MQTT Settings
> #define mqttServer IPAddress(192, 168, 0, 126)
> // const int mqttPort = 1883;
> const char *mqttUser = "MQTT-User";
> const char *mqttPassword = "***********";
> 
> // DHT11 Settings
> #define DHTPIN 4 // DHT11 data pin
> #define DHTTYPE DHT11
> DHT dht(DHTPIN, DHTTYPE);
> float dhtOffset = -5;
> float dhtLastCheck;
> float dhtCheckInterval = 1000;
> 
> // Home Assistant Settings
> // Entity Settings
> WiFiClient wifiClient;
> HADevice device("RetroStat-Basic");
> HAMqtt mqtt(wifiClient, device);
> HAHVAC climate(
>     "retrostat-hvac", HAHVAC::TargetTemperatureFeature);
> 
> HANumber humidity("RetroStat-Basic-Humidity");
> HANumber dhtOffsetNumber("RetroStat-Basic-DHT-Offset");
> 
> // Servo Settings
> Servo servo;
> const int servoPin = 7;
> 
> // Temperature Settings
> float currentTemperature = 20; // Initial temperature set to 20 degrees
> float targetTemperature = 20;
> const float minTemp = 10;
> const float maxTemp = 35;
> 
> // Function declarations
> void setServoPosition(int temp);
> void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender);
> void onDhtOffsetCommand(HANumeric value, HANumber *sender);
> 
> void setup()
> {
>     Serial.begin(115200);
>     dht.begin();
>     servo.attach(servoPin, 500, 2500);
> 
>     // Connect to WiFi
>     Serial.println("Connecting to WiFi...");
>     // WiFi.config(local_IP, gateway, subnet);
>     WiFi.begin(ssid, password);
>     while (WiFi.status() != WL_CONNECTED)
>     {
>         delay(500);
>         Serial.print(".");
>     }
>     Serial.println("WiFi connected");
> 
>     // Connect to MQTT
>     Serial.println("Connecting to MQTT...");
>     mqtt.begin(mqttServer, 1883, mqttUser, mqttPassword);
>     mqtt.setDataPrefix("homeassistant/climate"); // not fully sure about this one. defaults to "aha"
> 
>     // Initialize Home Assistant components
>     device.setName("RetroStat Basic");
>     device.setSoftwareVersion("2.0.0");
>     device.setManufacturer("MarshallVisions");
>     device.setModel("RS-Basic");
>     device.enableSharedAvailability();
>     device.enableLastWill();
>     climate.setName("RetroStat");
>     climate.setMinTemp(minTemp);
>     climate.setMaxTemp(maxTemp);
>     climate.setCurrentTemperature(currentTemperature);
>     climate.setTargetTemperature(targetTemperature);
>     climate.setRetain(true);
>     climate.onTargetTemperatureCommand(onTargetTemperatureCommand);
>     humidity.setName("Humidity");
>     humidity.setRetain(true);
>     dhtOffsetNumber.setName("DHT Temperature Offset");
>     dhtOffsetNumber.setRetain(true);
>     dhtOffsetNumber.onCommand(onDhtOffsetCommand);
> 
>     Serial.println("Setup complete.");
> }
> 
> void loop()
> {
>     mqtt.loop();
> 
>     if (millis() - dhtLastCheck > dhtCheckInterval)
>     {
>         // Read temperature from DHT11
>         float temp = dht.readTemperature() + dhtOffset;
>         float hum = dht.readHumidity();
>         currentTemperature = temp;
>         climate.setCurrentTemperature((float)currentTemperature);
>         humidity.setState(hum);
>         dhtLastCheck = millis();
>         Serial.print("Current temperature from DHT11: ");
>         Serial.println(temp);
>         Serial.print("Current humidity from DHT11: ");
>         Serial.println(hum);
>     }
> }
> 
> void setServoPosition(int temp)
> {
>     int pos = map(temp, minTemp, maxTemp, 180, 0);
>     Serial.print("Setting servo position to: ");
>     Serial.println(pos);
>     servo.write(pos);
> }
> 
> // Function to handle target temperature command from Home Assistant
> void onTargetTemperatureCommand(HANumeric temperature, HAHVAC *sender)
> {
>     float temperatureFloat = temperature.toFloat();
>     if (temperatureFloat >= minTemp && temperatureFloat <= maxTemp)
>     {
>         targetTemperature = temperatureFloat;
>         Serial.print("Target temperature command received. New target temperature: ");
>         Serial.println(temperatureFloat);
>         setServoPosition(targetTemperature);
>         sender->setTargetTemperature(temperature); // report target temperature back to the HA panel
>     }
> }
> 
> void onDhtOffsetCommand(HANumeric value, HANumber *sender)
> {
>     float offsetFloat = value.toFloat();
>     if (offsetFloat >= -10 && offsetFloat <= 10)
>     {
>         dhtOffset = offsetFloat;
>         Serial.print("DHT value command received. New value: ");
>         Serial.println(offsetFloat);
>         dhtLastCheck = millis();
>         sender->setState(dhtOffset);
>     }
> }
> ```

Probably this create the missing MQTT entries - this is what I found (mentioned above)
Thank you for sharing this, I will test it.

MarshallVisions, 2025-01-24:
can you so me a solid and edit out my wifi password in your reply lol.. I derped.

myriaxis, 2025-05-12:
@andrewhack , @MarshallVisions , @shotreal 
Hi, I'm having the same issue where I can't control the target temperature from HASS thermostat. I only have current temperature and mode selection.
I'm using an Arduino Nano ESP32 and coding on IDE 2.3.6 with all the latest libraries.
I compared MarshallVisions's code with mine and I don't see what I'm missing. What needs to be added to the Arduino code to add missing mqtt entries and make it work?
Also, shotreal suggested to publish something to the [...]/temp_cmd_t topic. How can I do that?

Your help would be greatly appreciated.

myriaxis, 2025-07-14:
@dawidchyrzynski  
I'm having the same issue where I can't control the target temperature from HASS thermostat. I only have current temperature and mode selection.
I am using the provided HVAC example with an Arduino Nano ESP32 and coding on IDE 2.3.6 with all the latest libraries.
Please help.