dawidchyrzynski/arduino-home-assistant #284: HASensorNumber - Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads

michelep, 2025-01-14:
Calling setValue on a HASensorNumber object cause Exception 28 on an ESP8266 board ESPduino 12E

Here's the piece of code, from a Dallas DS18B20 temperature sensor:

```
HADevice device;
HAMqtt mqtt(client, device);

HASensorNumber waterSensor("water_temp", HASensorNumber::PrecisionP2);

[...]

float waterTemp;

void loop() {
[...]
    sensors.requestTemperatures(); 
    waterTemp = sensors.getTempCByIndex(0);
    waterSensor.setValue(waterTemp);
  }
}

```

Here's the trace:

```
Decoding stack results
0x40212f13: HABaseDeviceType::publishOnDataTopic(__FlashStringHelper const*, unsigned char const*, unsigned short, bool, bool) at /Arduino/libraries/arduino-home-assistant/src/device-types/HABaseDeviceType.h line 44
0x40213250: HASensorNumber::onMqttConnected() at /Arduino/libraries/arduino-home-assistant/src/device-types/HABaseDeviceType.h line 44
0x40213226: HASensorNumber::publishValue(HANumeric const&) at /Arduino/libraries/arduino-home-assistant/src/device-types/HASensorNumber.cpp line 59
0x402132a8: HASwitch::buildSerializer() at /Arduino/libraries/arduino-home-assistant/src/device-types/HASwitch.cpp line 39
0x40202c00: loop() at ESP8266_Relay_X4/ESP8266_Relay_X4.ino line 431
0x40201d01: ArduinoJson::V730PB22::detail::parseNumber(char const*) at /Arduino/libraries/ArduinoJson/src/ArduinoJson/Numbers/FloatTraits.hpp line 204
0x40218648: user_rf_pre_init() at /.arduino15/packages/esp8266/hardware/esp8266/3.1.2/cores/esp8266/core_esp8266_phy.cpp line 362
```

romoloman, 2025-03-07:
The same problem exists also on esp32.
setValue on a HASensorNumber object causes crashes.

vpankin, 2025-03-15:
you need to set mac address to HAdevice
code from nodemcu example:
```
// Unique ID must be set!
    byte mac[WL_MAC_ADDR_LENGTH];
    WiFi.macAddress(mac);
    device.setUniqueId(mac, sizeof(mac));
```

romoloman, 2025-03-23:
It doesn't work even setting the uniqueId.
I did it in setup even declaring mac as static but it get not set.
Only if i declare device initialising it with a uniqueid it works, but indeed is not usable if you make several devices...

dawidchyrzynski, 2025-07-07:
@romoloman This crash was caused by a missing device ID. I suspect that you tried to push sensor data before setting the device’s ID. This has been fixed in version 2.2.0 - from now on, it will log a debug message instead of crashing.