dawidchyrzynski/arduino-home-assistant #287: Is there an API to send number values from HA to Arduino?

myriaxis, 2025-03-26:
I'd like to set a temperature value in HA and have Arduino react to this new value.
I couldn't find an example or description of a way to send number values from HA to Arduino.
Can you please help?
Thanks

cjbaar, 2025-05-07:
I think you are looking for the "HANumber" device type. You can define a callback in the Arduino code that gets run whenever HomeAssistant update the value via the MQTT state topic.
https://dawidchyrzynski.github.io/arduino-home-assistant/documents/api/device-types/ha-number.html

dawidchyrzynski, 2025-07-06:
@myriaxis There’s a HANumber class that already does exactly what you need. Check out this example:
https://github.com/dawidchyrzynski/arduino-home-assistant/blob/main/examples/number/number.ino

closed 2025-07-06 by dawidchyrzynski