dawidchyrzynski/arduino-home-assistant #293: Upgrading HA 2025.6.3 -> 2025.7.1 sensor stopped delivering data

zandzpider, 2025-07-06:
Hi!

I am not sure what information to give but my trusty esp32 temp sensor stopped reporting its temperature after upgrading HA.
I can see in the broker that it logs in and disconnects as normal, but no data is being delivered to the sensor.

Restarting HA including a full boot and also the sensor didn't resolve anything
Restoring back to the previous version made everything work again so something fishy is up

dawidchyrzynski, 2025-07-07:
Hi @zandzpider, I'll check this shortly.

Continuous2Discrete, 2025-07-16:
It’s possible that this is related to this issue. I noticed that display precision setting has stopped working in recent versions HA. Interestingly, the problem only affects HASensorNumber::PrecisionP1, where only an integer value is shown. In contrast, HASensorNumber::PrecisionP2 works correctly.

This might be related to this: https://github.com/esphome/issues/issues/7120

dawidchyrzynski, 2025-07-17:
Also, what state class or device class of the sensor are you using, if any?

zandzpider, 2025-08-05:
I was going to say it didn't work.

But then i added the sensor again via the MQTT button in HA and wrote in all the stuff i thought was right into the multistep add feature and it appeared with the data.
It's a shame that it just bricked my already existing MQTT devices that are ESP32/Arduino based.

Thank you for the help!

closed 2025-08-05 by zandzpider